adafruit/Adafruit_TSL2561 #2: stops working after a while

xseignard, 2015-01-25:
Hello,

I don't know why, but the sensor stops working (i.e. no more event.light == true) after a while.

Here is my sketch: https://gist.github.com/xseignard/2d4e00fd3a762d393606

The sketch still runs, but no more lux measures,

Any idea?

Regards,

Xavier

xseignard, 2015-01-28:
So the problem is that, as opposed as it is written in the example, the sensor can actually return 0 lux. So if the sensors returns 0 it is not sure that it's a sensor overload.

Beeski, 2015-02-11:
I experienced this as well. So I cached the last known non-zero value and returned that if the lux reading was zero. That way if the sensor suddenly finds it self in full sun then it just plateaus along until it gets another valid reading and doesn't just instantly drop to zero.   

HTH

beegee-tokyo, 2016-01-13:
I made a change to the library to get around this zero value problem. See my pull request
https://github.com/adafruit/Adafruit_TSL2561/pull/3
The changed library is as well available in my fork 
https://github.com/beegee-tokyo/Adafruit_TSL2561

ladyada, 2018-01-17:
oops this was merged and we didnt close. thanks!

closed 2018-01-17 by ladyada