adafruit/Adafruit_TSL2561 #13: Init() Fix #11 only half fixed the problem

sethtroisi, 2018-08-06:
In the fix for #11, https://github.com/adafruit/Adafruit_TSL2561/commit/c913785d70a63774f7881bb1cbd072cb191eb94f#diff-fb32052893c4375234c2bcd52e31e8edR110 `if (x & 0xF0 != 0x10) {` was added to `::Init()`

This is equivalent to `if (x & 1) {`
I believe the desired code is to add more parenthesis, e.g. `if ((x & 0xF0) != 0x10) {

I tested with this simple program:
`
#include<iostream>
using namespace std;
int main() {
    for (int x = 0; x < 128; x++) {
        bool current = (x & 0xF0 != 0x10);
        bool test    = ((x & 0xF0) != 0x10);
        cout << x << " " << current << " " << test << endl;
    }
}
`

I'm not sure this actually fixes the problem because my TSL2561 (ordered in 2016) returns 0x0A similiar to @beegee-tokyo.

sethtroisi, 2018-08-06:
@ladyada

sethtroisi, 2018-08-06:
Continuous integration (for my code beautification PR) caught this as an issue :)

[image]

ladyada, 2018-08-06:
ok plz update the PR!

yoursunny, 2019-03-01:
#17 had this fix merged, but it caused my device to stop working.
I'm using ESP32 (Heltec\_WiFi\_Kit\_32) and [Adafruit TSL2561 breakout](https://www.adafruit.com/product/439) (shipped Sep 2018).

Using an older commit 0558d76ba491b25b773f80c0d6946a05ce25c466 , the sensor is detected and I can retrieve reasonable readings.
Using current commit 4f670c0cffb460a341f500433398f7bd0117db62 , sensor initialization fails with *Ooops, no TSL2561 detected ... Check your wiring or I2C ADDR!*. I inserted logging to `Adafruit_TSL2561_Unified::init` function and noticed that `x` equals `0x0A`.
I know the datasheet says upper nibble of the ID register should be 0x1 for TSL2561 but this isn't the case in reality.

ladyada, 2019-03-12:
@yoursunny heya! could you try to add an extra boolean test on that line in case the address is 0x0A - we dont know whats up with this but it seems some TSL's have different ID codes for some mysterious reason! then submit a new PR :)

ladyada, 2019-04-03:
i think this is resolved. well, i hope it is!

closed 2019-04-03 by ladyada