adafruit/Adafruit_TSL2561 #21: Which Adafruit TSL2561 library should I use?

kankaristo, 2019-05-03:
I just noticed that Adafruit has a second library for using the TSL2561:
https://github.com/adafruit/TSL2561-Arduino-Library

The other library is clearly older (when you look at the commits), but neither library's readme mentions the other library, so I'm not sure which library I should use... The older library was the first Google hit for me, so other people might find that one first, too.

If the older library is no longer being developed, maybe a note should be added to its readme? I think I remember seeing "archived" repositories on GitHub, so that's a possibility, if the older library should no longer be used.

Is the older library deprecated? What are the differences between the libraries, is there some reason why I might want to use the older library instead of this one?

ladyada, 2019-05-03:
yes please use this one, its maintained more and uses Adafruit Sensor

caternuson, 2019-05-07:
OK - other library marked as deprecated and archived. Thanks for pointing this out. Hopefully that clears things up. Will also send notice to have it removed from Arduino Library Manager and over time the Google will start favoring this one.

closed 2019-05-07 by caternuson