adafruit/Adafruit_TSL2561 #22: getting 65536.00 value from event.light

TinajaLabs, 2019-05-04:
I'm building a room sensor with DHT22, a TSL2561, and an Adafruit PIR sensor (not yet hooked up).  These are being plugged into a wemos d1 mini with wifiManager and MQTT libs included.

When I run the simple sensorapi application, unadjusted, I get a reasonable value for the lux reading.

When I run it with my fairly complicated code, I get 65536.

lux: 65536.00
broadband:	65535	
infrared:	65535

I added the `getLuminosity()` call to see what they were showing.  Here's the code that does the reading.
```
  /* Get a new sensor event */
  sensors_event_t event;
  tsl.getEvent(&event);

  tsl.getLuminosity(&broadband, &infrared);

  /* Display the results (light is measured in lux) */
  if (event.light)
  {
    Serial.print(event.light);
    Serial.print(" lux\t");

    Serial.print("broadband:\t");
    Serial.print(broadband);

    Serial.print("\tinfrared:\t");
    Serial.println(infrared);

  }
  else
  {
    /* If event.light = 0 lux the sensor is probably saturated
       and no reliable data could be generated! */
    Serial.println("Sensor overload");
  }
  delay(250);
```

At the top of the sketch, I'm loading the following libs:
```
#include <FS.h> //this needs to be first, or it all crashes and burns...

#include <Wire.h>
#include <Adafruit_Sensor.h>  // Adafruit Unified Sensor Driver - https://github.com/adafruit/Adafruit_Sensor

#include <ESP8266WiFi.h> //https://github.com/esp8266/Arduino
#include <WiFiManager.h>      //https://github.com/tzapu/WiFiManager
#include <ESP8266WebServer.h> //Local WebServer used to serve the configuration portal
#include <ArduinoJson.h>      //https://github.com/bblanchon/ArduinoJson

#include <PubSubClient.h>
#ifdef ESP32
#include <SPIFFS.h>
#endif

#include <Adafruit_TSL2561_U.h>
#include <DHT.h>
```

It all compiles just fine.  Perhaps there are some conflicts with these libs?  

Thanks for any advice,
Chris.

ladyada, 2019-05-04:
probably a wiring issue. if the TSL2561 breakout is made and sold by adafruit you can post in the adafruit support forums. Otherwise, please post in the vendor support forums

closed 2019-05-04 by ladyada

TinajaLabs, 2019-05-04:
Yes it is the Adafruit TSL2561 Luminosity/Lux/Light Sensor
adafru.it/439

I'll repeat the question on the adafruit forum.

For others who follow, the reason I'm confused is that using the basic example sketch, sensorapi, it works.  It's only when I use it with other components and libraries that it fails.  So I feel sure it's not a wiring problem.

Thanks.